NVIDIA/tegra-nouveau-rootfs #3: [Boot error] "Unsupported resolution" appears on screen

geelen, 2015-01-26:
Hi, just went through and followed the instructions to get a Nouveau rootfs on an SD card. The uboot that was on the device already after flashing with L4T 21 worked straight away, which was good. But I get nothing on the screen, just a message saying "Unsupported resolution" (I'm using an old TV that works ok under the stock L4T, don't have anything else on hand)

I've attached [the full bootlog from the serial port](https://gist.github.com/geelen/ffb625655e84a2f28e1b) but I'm pretty new to all this so I can't figure out where to go from here.

Running `kmscube` causes the screen to flash, but again I get a message "Unsupported resolution". This is the output on the console:

```
ubuntu@tegra-ubuntu:~$ kmscube /dev/dri/card0 /dev/dri/renderD128
opening /dev/dri/card0 for drm
opening /dev/dri/renderD128 for gbm
MESA-LOADER: malformed or no PCI ID
MESA-LOADER: malformed or no PCI ID
Using display 0x3c618 with EGL version 1.4
EGL Version "1.4 (DRI2)"
EGL Vendor "Mesa Project"
EGL Extensions "EGL_MESA_drm_image EGL_MESA_configless_context EGL_WL_bind_wayland_display EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_gl_renderbuffer_image EGL_KHR_surfaceless_context EGL_KHR_create_context EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import "
[  698.492389] tegra-hdmi 54280000.hdmi: cannot set audio to 44100 at 119000000 pclk
^C[  741.416090] tegra-hdmi 54280000.hdmi: cannot set audio to 44100 at 78800000 pclk
```

Any advice how to check what display devices are being detected & maybe force them into a certain resolution/refresh rate?

Gnurou, 2015-01-26:
Hi, thanks for reporting this. Looking at the logs it seems like tegradrm cannot find any valid mode for your display. Assigned to @thierryreding who wrote this driver and may understand why this is so.

geelen, 2015-01-26:
Thanks! I'm holed up in a hotel room trying to hack on my Tegra with this crappy old TV, so I don't have any other screen to test with. If it helps, booting the L4T base rom has no problems detecting the TV as a 1360x768 screen, but my mac and my other device (an Odroid U3) both can only send down 1280x720. 

Also, for the U3, it worked best when I stopped it looking for an EDID and fixed it to 1024x768 "dvi mode" ([relevant forum post](http://forum.odroid.com/viewtopic.php?f=52&t=2947)), which is what I was hoping to do here.

Thanks for your help!

geelen, 2015-01-30:
Went and bought a real screen, it boots up fine! Would be nice to have a workaround for people in the future if they hit this issue, though.

Gnurou, 2015-01-30:
Yeah, it's quite a corner case but I am keeping this issue open in case @thierryreding can give it a look.

Gnurou, 2015-10-17:
Closing this for now.

closed 2015-10-17 by Gnurou