NVIDIA/tegra-nouveau-rootfs #6: No hdmi outpout

gallit, 2015-05-20:
Hello,
I followed the process and i have the following result :
  - boot ok
  - access ok from serial
  - no hdmi output

here is an extract of dmesg :

ubuntu@tegra-ubuntu:~$ dmesg|grep hdmi
[    1.208327] tegra-hdmi 54280000.hdmi: failed to get HDMI regulator
[    1.214563] platform 54280000.hdmi: Driver tegra-hdmi requests probe deferral
[    2.380516] tegra-hdmi 54280000.hdmi: failed to get PLL regulator
[    2.386708] platform 54280000.hdmi: Driver tegra-hdmi requests probe deferral
ubuntu@tegra-ubuntu:~$ 

I had the issue just after first boot and after upgrades.

Gnurou, 2015-05-21:
Hi, could you post the entire dmesg log? These first messages are very early in the kernel and I think they are expected at that time. I'd like to see if there is something else.

Also, have you tried with a different HDMI monitor?

Gnurou, 2015-05-21:
FWIW I have just tried rebuilding everything from scratch and booting, and HDMI was working. A complete dmesg would help understanding what is going wrong here.

gallit, 2015-05-21:
Hello, i'm sorry, i changed the monitor and it's working.
The strange thing is that the monitor is working with a different computer...
I now have console login but no graphical ui.
I built everything, including wayland/weston.

Here is the full dmesg :
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.0.0-171721-g12c466d (gallit@vm1) (gcc version 4.8.3 20140401 (prerelease) (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04) ) #1 SMP PREEMPT Tue May 19 17:13:27 CEST 2015
[    0.000000] CPU: ARMv7 Processor [413fc0f3] revision 3 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine model: NVIDIA Tegra124 Jetson TK1
[    0.000000] cma: Reserved 64 MiB at 0xfbc00000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 524032
[    0.000000] free_area_init_node: node 0, pgdat c0967e80, node_mem_map ee7f7000
[    0.000000]   Normal zone: 1520 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 194560 pages, LIFO batch:31
[    0.000000]   HighMem zone: 329472 pages, LIFO batch:31
[    0.000000] PERCPU: Embedded 11 pages/cpu @ee799000 s12672 r8192 d24192 u45056
[    0.000000] pcpu-alloc: s12672 r8192 d24192 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 522512
[    0.000000] Kernel command line: console=ttyS0,115200n8 console=tty1 rw rootwait root=PARTUUID=1d643aff-bdcb-4a82-b85c-8f2146b2b6e1 nouveau.pstate=1 text
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 2002928K/2096128K available (6711K kernel code, 318K rwdata, 2268K rodata, 304K init, 415K bss, 27664K reserved, 65536K cma-reserved, 1252352K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc08cd080   (8981 kB)
[    0.000000]       .init : 0xc08ce000 - 0xc091a000   ( 304 kB)
[    0.000000]       .data : 0xc091a000 - 0xc0969960   ( 319 kB)
[    0.000000]        .bss : 0xc0969960 - 0xc09d1570   ( 416 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] L2C: failed to init: -19
[    0.000009] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000032] Switching to timer-based delay loop, resolution 1000ns
[    0.001168] Architected cp15 timer(s) running at 12.00MHz (virt).
[    0.001193] sched_clock: 56 bits at 12MHz, resolution 83ns, wraps every 2863311536128ns
[    0.001206] Switching to timer-based delay loop, resolution 83ns
[    0.001605] Console: colour dummy device 80x30
[    0.002168] console [tty1] enabled
[    0.002203] Calibrating delay loop (skipped), value calculated using timer frequency.. 24.00 BogoMIPS (lpj=120000)
[    0.002248] pid_max: default: 32768 minimum: 301
[    0.002663] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002694] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.004084] Initializing cgroup subsys freezer
[    0.004127] Initializing cgroup subsys debug
[    0.004197] CPU: Testing write buffer coherency: ok
[    0.004514] /cpus/cpu@0 missing clock-frequency property
[    0.004567] /cpus/cpu@1 missing clock-frequency property
[    0.004618] /cpus/cpu@2 missing clock-frequency property
[    0.004671] /cpus/cpu@3 missing clock-frequency property
[    0.004700] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004761] Setting up static identity map for 0x80660220 - 0x806602b8
[    0.041333] Tegra Revision: A01 SKU: 129 CPU Process: 1 Core Process: 1
[    0.061770] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.081770] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.101782] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.101875] Brought up 4 CPUs
[... 349 lines cut ...]
[   13.629751] nouveau  [  DEVICE][57000000.gpu] Family : NVE0
[   13.666428] nouveau  [ INSTMEM][57000000.gpu] using IOMMU
[   13.667449] nouveau  [    VOLT][57000000.gpu] The default voltage is 1000000uV
[   13.667476] nouveau  [    VOLT][57000000.gpu] GPU voltage: 1035000uv
[   13.667501] nouveau  [     CLK][57000000.gpu] parent clock rate: 12 Mhz
[   13.669321] nouveau  [     CLK][57000000.gpu] --: core 198 MHz  
[   13.814173] nouveau  [  PGRAPH][57000000.gpu] using external firmware
[   13.960217] [TTM] Zone  kernel: Available graphics memory: 375440 kiB
[   13.960243] [TTM] Zone highmem: Available graphics memory: 1034384 kiB
[   13.960264] [TTM] Initializing pool allocator
[   13.960382] [TTM] Initializing DMA pool allocator
[   13.960574] nouveau  [     DRM] VRAM: 0 MiB
[... cut ...]
[   14.600539] r8169 0000:01:00.0: Direct firmware load for rtl_nic/rtl8168g-2.fw failed with error -2
[   14.600566] r8169 0000:01:00.0 eth0: unable to load firmware patch rtl_nic/rtl8168g-2.fw (-2)
[   14.617739] r8169 0000:01:00.0 eth0: link down
[   14.617772] r8169 0000:01:00.0 eth0: link down
[   14.617901] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   14.912033] init: plymouth-stop pre-start process (839) terminated with status 1
[   16.256741] r8169 0000:01:00.0 eth0: link up
[   16.256772] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

gallit, 2015-05-21:
And the Xorg.0.log :

[   105.190] 
X.Org X Server 1.15.1
Release Date: 2014-04-13
[   105.204] X Protocol Version 11, Revision 0
[   105.209] Build Operating System: Linux 3.2.0-67-highbank armv7l Ubuntu
[   105.214] Current Operating System: Linux tegra-ubuntu 4.0.0-171721-g12c466d #1 SMP PREEMPT Tue May 19 17:13:27 CEST 2015 armv7l
[   105.214] Kernel command line: console=ttyS0,115200n8 console=tty1 rw rootwait root=PARTUUID=1d643aff-bdcb-4a82-b85c-8f2146b2b6e1 nouveau.pstate=1 text
[   105.224] Build Date: 12 February 2015  02:55:07PM
[   105.229] xorg-server 2:1.15.1-0ubuntu2.7 (For technical support please see http://www.ubuntu.com/support) 
[   105.234] Current version of pixman: 0.30.2
[   105.244] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   105.244] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   105.264] (==) Log file: "/var/log/Xorg.0.log", Time: Wed May 20 11:13:37 2015
[   105.284] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   105.300] (==) No Layout section.  Using the first Screen section.
[   105.300] (==) No screen section available. Using defaults.
[   105.300] (**) |-->Screen "Default Screen Section" (0)
[   105.300] (**) |   |-->Monitor "<default monitor>"
[   105.301] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[   105.302] (==) Automatically adding devices
[   105.302] (==) Automatically enabling devices
[   105.302] (==) Automatically adding GPU devices
[   105.319] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   105.319] 	Entry deleted from font path.
[   105.319] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   105.319] 	Entry deleted from font path.
[   105.319] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   105.319] 	Entry deleted from font path.
[   105.320] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   105.321] 	Entry deleted from font path.
[   105.321] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   105.321] 	Entry deleted from font path.
[   105.321] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/Type1,
	built-ins
[   105.321] (==) ModulePath set to "/usr/lib/arm-linux-gnueabihf/xorg/extra-modules,/usr/lib/xorg/extra-modules,/usr/lib/xorg/modules"
[   105.321] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   105.321] (II) Loader magic: 0xb6fb1f10
[   105.321] (II) Module ABI versions:
[   105.321] 	X.Org ANSI C Emulation: 0.4
[   105.321] 	X.Org Video Driver: 15.0
[   105.321] 	X.Org XInput driver : 20.0
[   105.321] 	X.Org Server Extension : 8.0
[   105.323] (II) xfree86: Adding drm device (/dev/dri/card0)
[   105.323] (II) xfree86: Adding drm device (/dev/dri/card1)
[   105.324] (EE) 
[   105.329] (EE) Backtrace:
[   105.334] (EE) 
[   105.339] (EE) Segmentation fault at address 0x0
[   105.344] (EE) 
Fatal server error:
[   105.353] (EE) Caught signal 11 (Segmentation fault). Server aborting
[   105.358] (EE) 
[   105.363] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   105.382] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   105.387] (EE)

Gnurou, 2015-05-23:
Is this the dmesg with the failed monitor? I'd like to see this one to understand why this particular monitor doesn't work.

Regarding X - X is not supported yet, although Marteen Lankhorst has a couple of patches to add support (don't know how well it will work though). You can use Weston as explained on https://github.com/NVIDIA/tegra-nouveau-rootfs .

gallit, 2015-05-24:
OK thanks for X.

About the failed monitor, yes, this is that dmesg.
I finally had it working by unplugging the VGA and usb connections and booting again.

The monitor is an LG one. I'll give you the exact model if you want. It has VGA and HDMI.

I have my host computer sharing the monitor while i had serial ans USB connections between jetson and x86 host.

The monitor says 'no HDMI signal' if i boot with all the connections.

I use an USB disk for booting jetson, on a USB hub.

It's a strange issue but i can close as it's not a jetson issue.
If you still want to investigate, i can give you more info. Just tell me.

Thanks again.

closed 2015-05-29 by gallit