NVIDIA/tegra-nouveau-rootfs #10: nouveau firmware load error

gallit, 2015-09-02:
I'm having this error when i try kmscube or weston-launch :
[  202.248149] nouveau 57000000.gpu: Direct firmware load for nvidia/gk20a/fecs_inst.bin failed with error -2

I cann't find that file in the rootfs.

Here is full dmesg (ubuntu core rootfs)
http://pastebin.com/PqLgVFcT

I tried with the default rootfs and had the same.

Gnurou, 2015-09-03:
Have you properly run ./scripts/install-firmware? This should put the required firmware files in /lib/firmware. E.g. check for /lib/firmware/nvidia/gk20a/fecs_inst.bin , it should be there or Nouveau will not load.

gallit, 2015-09-03:
Yes i did.
I have a /lib/firmware/nouveau folder, not nvidia.
And that file is not there.

Here is the content of the nouveau folder :
nv12b_acr_ucode
nv12b_fecs_bl
nv12b_fecs_sig
nv12b_pmu_bl

Gnurou, 2015-09-03:
I think you need to repo sync the linux-firmware repository and run ./scripts/install-firmware again - you have probably ran it against an old version. Make sure that /lib/firmware/nvidia/gk20a/ exists and is well-populated.

gallit, 2015-09-03:
i do have the nvidia folder in the linux-firmware repo.

i checked the install-firmware script.
it does a copy the nouveau folder only, not the nvidia.

cp -R $TOP/$PACKAGE/nouveau $SYSROOT/lib/firmware/

Gnurou, 2015-09-03:
Oh wait, install-firmware appears to be broken... Let me fix this.

Gnurou, 2015-09-03:
There, it should be fixed - could you repo sync and try to run install-firmware again?

gallit, 2015-09-03:
Ok, thanks.

Gnurou, 2015-09-03:
Confirmed that things should be working fine now - please reopen if you are still having issues, and thanks for reporting!

closed 2015-09-03 by Gnurou