NVIDIA/tegra-nouveau-rootfs #14: Wayland scripts are gone

gallit, 2015-10-13:
Hello,
It seems that you removed wayland.

Gnurou, 2015-10-13:
Hi,

The default distro to build has switched to Arch Linux (Ubuntu had too many outdated packages to make recompilation practical, and Arch is just simpler to build generally speaking). The scripts to build L4T are still here, but I will not support this distro anymore and you will have to uncomment the missing repos in the manifest.

If Arch Linux is ok for you, please follow the instructions from the start again and you will have a running system in no time (and X comes as a bonus too!)

Gnurou, 2015-10-13:
I should also have mentioned that Arch's Wayland package is sufficiently up-to-date that we don't need to compile our own. You will still need to compile Weston though.

Closing for now, feel free to reopen if there is a need.

closed 2015-10-13 by Gnurou