NVIDIA/tegra-nouveau-rootfs #15: kmscube doesn't build cleanly with gcc-5

wom-bat, 2015-10-15:
```
function ‘drm_tegra_import’:
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/kmscube/kmscube.c:98:8: warning: implicit declaration of function ‘ioctl’ [-Wimplicit-function-declaration]
  err = ioctl(drm.fd, DRM_IOCTL_TEGRA_GEM_SET_TILING, &args);
        ^
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/kmscube/kmscube.c: In function ‘drm_fb_get_from_bo’:
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/kmscube/kmscube.c:641:4: warning: implicit declaration of function ‘close’ [-Wimplicit-function-declaration]
    close(fd);
    ^
  CC       kmscube-esTransform.o
  CCLD     kmscube
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/out/target/ArchLinuxArm/lib/librt.so.1: undefined reference to `__socket@GLIBC_PRIVATE'
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/out/target/ArchLinuxArm/lib/librt.so.1: undefined reference to `__pthread_barrier_init@GLIBC_PRIVATE'
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/out/target/ArchLinuxArm/lib/librt.so.1: undefined reference to `__pthread_barrier_wait@GLIBC_PRIVATE'
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/out/target/ArchLinuxArm/lib/librt.so.1: undefined reference to `__libc_pread@GLIBC_PRIVATE'
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/out/target/ArchLinuxArm/lib/librt.so.1: undefined reference to `__shm_directory@GLIBC_PRIVATE'
/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/out/target/ArchLinuxArm/lib/librt.so.1: undefined reference to `__recv@GLIBC_PRIVATE'
collect2: error: ld returned 1 exit status
Makefile:490: recipe for target 'kmscube' failed
make[1]: *** [kmscube] Error 1
make[1]: Leaving directory '/home/peterc/src/work/scratch/jetson/tegra-nouveau-rootfs/out/build/ArchLinuxArm/kmscube'
Makefile:387: recipe for target 'all' failed
make: *** [all] Error 2
```

wom-bat, 2015-10-15:
Explicitly adding -L..../out/target/ArchLinuxArm/lib allowed building to complete.

wom-bat, 2015-10-15:
And the same problem arises when building weston.

Gnurou, 2015-10-15:
The version is gcc used should be the one downloaded by download-gcc - and it should work fine. What makes you think gcc 5 is being used here?

wom-bat, 2015-10-15:
>>>>> "Alexandre" == Alexandre Courbot <notifications@github.com> writes:

Alexandre> The version is gcc used should be the one downloaded by
Alexandre> download-gcc - and it should work fine. What makes you
Alexandre> think gcc 5 is being used here?

Simple -- I didn't use the download-gcc script, I used the
arm-linux-gnueabihf-gcc-5 I already had on the system.

So you can regard this as a heads-up for when you switch to a later
compiler.

-- 
Dr Peter Chubb				         http://www.data61.csiro.au
http://www.ssrg.nicta.com.au   Software Systems Research Group/NICTA/Data61

Gnurou, 2015-10-17:
Indeed, we will want to fix that sometime. Thanks for reporting, keeping open until this is fixed.

winsock, 2015-12-18:
I just want to say I wasn't able to reproduce this issue with gcc5.3 cross compiled to arm-linux-gnueabihf.
I'm running on a Gentoo amd64 VM. Only script modified was env.sh to make CROSS_COMPILE point to my cross-compiler.

Gnurou, 2015-12-18:
closing for now.

closed 2015-12-18 by Gnurou