NVIDIA/tegra-nouveau-rootfs #16: prepare-rootfs failure

gallit, 2015-10-15:
Hi,
I'm trying to build from is a fresh archlinux host...

I first noticed a failure on the download-gcc script (wayland/autogen.sh is not here)
# wayland-scanner
if [ ! -f "$TOP/$LINARO_GCC_PACKAGE/bin/wayland-scanner" ]; then
    mkdir -p $TOP/out/build/wayland-scanner
    cd $TOP/out/build/wayland-scanner
    $TOP/wayland/autogen.sh --disable-documentation
    make wayland-scanner
    cp wayland-scanner $TOP/$LINARO_GCC_PACKAGE/bin
fi

As it's not a serious issue, i continued and it fails on the prepare-rootfs script.

[gallit@archi nouveau]$ 
[gallit@archi nouveau]$ ./scripts/prepare-rootfs
Installing qemu-arm-static binary into target FS... 
--2015-10-15 14:53:51--  http://snapshot.debian.org/archive/debian/20141030T220503Z/pool/main/q/qemu/qemu-user-static_2.1+dfsg-5+b1_i386.deb
Resolving snapshot.debian.org (snapshot.debian.org)... 193.62.202.30, 185.17.185.187, 2001:630:206:4000:1a1a:0:c13e:ca1e, ...
Connecting to snapshot.debian.org (snapshot.debian.org)|193.62.202.30|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5882238 (5.6M) [application/x-debian-package]
Saving to: '/tmp/tmp.oL3jH5aF9s'

/tmp/tmp.oL3jH5aF9s                     100%[==============================================================================>]   5.61M  2.05MB/s   in 2.7s   

2015-10-15 14:53:54 (2.05 MB/s) - '/tmp/tmp.oL3jH5aF9s' saved [5882238/5882238]

./scripts/prepare-rootfs: line 49: /home/gallit/dev/nouveau/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux/bin/arm-linux-gnueabihf-ar: No such file or directory
xz: (stdin): File format not recognized
tar: Child returned status 1
tar: Error is not recoverable: exiting now
[gallit@archi nouveau]$

Gnurou, 2015-10-15:
The download-gcc script failed and as a result you did not have the ar binary. The failure is because I overlooked the fact that wayland is still needed for that step. I will fix this tomorrow and come back to you. Nice catch.

Gnurou, 2015-10-17:
Ok, this should be fixed now - could you confirm after doing a repo sync and running from "download-gcc" again?

gallit, 2015-10-20:
Hi, 
I'm having some troubles with repo, so i cannot try for now.
I'm looking for a tip to make python 2.7 as default.

Feel free to close this issue, i'll repost if needed.
:(

[gallit@archi jetson]$ repo init -u https://github.com/NVIDIA/tegra-nouveau-rootfs.git -m tegra-nouveau.xml
warning: Python 3 support is currently experimental. YMMV.
Please use Python 2.6 - 2.7 instead.
Traceback (most recent call last):
  File "/usr/bin/repo", line 860, in <module>
    main(sys.argv[1:])
  File "/usr/bin/repo", line 828, in main
    _Init(args, gitc_init=(cmd == 'gitc-init'))
  File "/usr/bin/repo", line 326, in _Init
    _CheckGitVersion()
  File "/usr/bin/repo", line 380, in _CheckGitVersion
    ver_act = ParseGitVersion(ver_str)
  File "/usr/bin/repo", line 350, in ParseGitVersion
    if not ver_str.startswith('git version '):
TypeError: startswith first arg must be bytes or a tuple of bytes, not str

Gnurou, 2015-10-20:
sudo ln -sf /usr/bin/python2 /usr/bin/python ?

gallit, 2015-10-20:
Damned ! So simple...
I was looking for uninstall v3 etc...

gallit, 2015-10-20:
The issue is still here :

[gallit@archi jetson]$ ./scripts/prepare-rootfs
Installing qemu-arm-static binary into target FS... 
--2015-10-20 11:38:41--  http://snapshot.debian.org/archive/debian/20141030T220503Z/pool/main/q/qemu/qemu-user-static_2.1+dfsg-5+b1_i386.deb
Resolving snapshot.debian.org (snapshot.debian.org)... 193.62.202.30, 185.17.185.187, 2001:1af8:4020:b030:deb::187, ...
Connecting to snapshot.debian.org (snapshot.debian.org)|193.62.202.30|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-debian-package]
Saving to: '/tmp/tmp.VWULkx1TOx'

/tmp/tmp.VWULkx1TOx                         [               <=>                                                             ]   5.61M  1.93MB/s   in 2.9s   

2015-10-20 11:38:47 (1.93 MB/s) - '/tmp/tmp.VWULkx1TOx' saved [5882238]

./scripts/prepare-rootfs: line 49: /home/gallit/dev/jetson/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux/bin/arm-linux-gnueabihf-ar: No such file or directory
xz: (stdin): File format not recognized
tar: Child returned status 1
tar: Error is not recoverable: exiting now
[gallit@archi jetson]$

Gnurou, 2015-10-27:
Please remove /home/gallit/dev/jetson/gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux and run download-gcc again - it should succeed, as should prepare-rootfs.

gallit, 2015-11-03:
Well, i removed the gcc folder and retried.
No success.

Here is the full log: http://pastebin.com/QuSs8ELa

gallit, 2015-11-06:
Hello,
From today, the prepare-rootfs script is working !
I noticed that you updated gcc to gcc-linaro-4.8-2015.06-x86_64_arm-linux-gnueabihf.
The build is in progress.

Thanks.

closed 2015-11-06 by gallit

Gnurou, 2015-11-06:
Ah well - I don't know what happened, but good news.

Yes, the toolchain has been updated in prevision of supporting 64-bit userspaces. :)