NVIDIA/tegra-nouveau-rootfs #17: ./scripts/build-xserver issue

gallit, 2015-11-06:
It's me again...
I'm now having issue with the build XServer script.
Can you tell me if it's working on your side ?

[gallit@archi jetson]$ ./scripts/build-xserver
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "en_US.UTF-8",
	LC_MONETARY = "en_US.UTF-8",
	LC_NUMERIC = "en_US.UTF-8",
	LC_MEASUREMENT = "en_US.UTF-8",
	LC_TIME = "en_US.UTF-8",
	LANG = "en_US"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "en_US.UTF-8",
	LC_MONETARY = "en_US.UTF-8",
	LC_NUMERIC = "en_US.UTF-8",
	LC_MEASUREMENT = "en_US.UTF-8",
	LC_TIME = "en_US.UTF-8",
	LANG = "en_US"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "en_US.UTF-8",
	LC_MONETARY = "en_US.UTF-8",
	LC_NUMERIC = "en_US.UTF-8",
	LC_MEASUREMENT = "en_US.UTF-8",
	LC_TIME = "en_US.UTF-8",
	LANG = "en_US"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "en_US.UTF-8",
	LC_MONETARY = "en_US.UTF-8",
	LC_NUMERIC = "en_US.UTF-8",
	LC_MEASUREMENT = "en_US.UTF-8",
	LC_TIME = "en_US.UTF-8",
	LANG = "en_US"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "en_US.UTF-8",
	LC_MONETARY = "en_US.UTF-8",
	LC_NUMERIC = "en_US.UTF-8",
	LC_MEASUREMENT = "en_US.UTF-8",
	LC_TIME = "en_US.UTF-8",
	LANG = "en_US"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "en_US.UTF-8",
	LC_MONETARY = "en_US.UTF-8",
	LC_NUMERIC = "en_US.UTF-8",
	LC_MEASUREMENT = "en_US.UTF-8",
	LC_TIME = "en_US.UTF-8",
	LANG = "en_US"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_PAPER = "en_US.UTF-8",
	LC_MONETARY = "en_US.UTF-8",
	LC_NUMERIC = "en_US.UTF-8",
	LC_MEASUREMENT = "en_US.UTF-8",
	LC_TIME = "en_US.UTF-8",
	LANG = "en_US"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
autoreconf: running: /usr/bin/autoconf --force
configure.ac:660: error: possibly undefined macro: XTRANS_CONNECTION_FLAGS
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
[gallit@archi jetson]$

Gnurou, 2015-11-10:
It is working for me, yes. What is your host distribution?

gallit, 2015-11-10:
Linux archi 4.2.5-1-ARCH #1 SMP PREEMPT Tue Oct 27 08:13:28 CET 2015 x86_64 GNU/Linux
Should be a 32 bits host ?

Gnurou, 2015-11-10:
x86_64 -> it's 64 bits

And what is your distribution?

gallit, 2015-11-10:
ArchLinux

Gnurou, 2015-11-10:
Uh. Same as me, that's interesting. Let me look in more detail.

gallit, 2015-11-10:
Sorry for my english (maybe we could talk in french...).
I know it's a 64 bits, i wondered if 32 bits was a required.
OK, thanks for the help.

Gnurou, 2015-11-11:
Can you try installing the "xtrans" package on your host and try again after removing out/build/arm/ArchLinuxArm/xserver ? Normally this package is installed on the guest FS and it should be enough, but maybe the script is not looking at the right place...

Gnurou, 2015-11-12:
Ok, I managed to reproduce your error by removing the xtrans package from my host, so installing it on your end should work as a workaround. I don't think it should be needed though, so keeping this issue open until I get to the bottom of it.

Gnurou, 2015-11-12:
Found it - ACLOCAL_PATH needed to be defined. I have updated the scripts, please repo sync and re-try to build xserver, it should pass without any problem now. Thanks for reporting this!

closed 2015-11-12 by Gnurou

gallit, 2015-11-12:
Ok, thanks for the support.