NVIDIA/tegra-nouveau-rootfs #18: jetson boot failed

gallit, 2015-11-12:
The board doesn't boot.
The boot script tries to boot an arm64 kernel.
I looked the boot-nouveau.scr.txt and there is a test on top of it.
I don't have any test environment variable so i don't understand why it enters on the armv8 bloc.

My uboot is : 
U-Boot 2015.10-rc5-00002-gf861f51 (Oct 13 2015 - 11:41:41 +0200)
arm-linux-gnueabi-gcc (Ubuntu/Linaro 4.7.3-12ubuntu1) 4.7.3
GNU ld (GNU Binutils for Ubuntu) 2.24

Here is the log:

Tegra124 (Jetson TK1) # boot
MMC: no card present
switch to partitions #0, OK
mmc0(part 0) is current device
** Invalid partition 1 **
starting USB...
USB0:   USB EHCI 1.10
USB1:   USB EHCI 1.10
scanning bus 0 for devices... 1 USB Device(s) found
scanning bus 1 for devices... 6 USB Device(s) found
       scanning usb for ethernet devices... 0 Ethernet Device(s) found

USB device 0:
    Device 0: Vendor: Generic  Rev: 9.02 Prod: USB Disk
            Type: Hard Disk
            Capacity: 61057.0 MB = 59.6 GB (125044736 x 512)
... is now current device
Scanning usb 0:1...
Found U-Boot script /boot/boot.scr
719 bytes read in 226 ms (2.9 KiB/s)
## Executing script at 90000000
** File not found /boot/Image-nouveau-arm64 **
64138 bytes read in 185 ms (337.9 KiB/s)
Unknown command 'booti' - try 'help'
SCRIPT FAILED: continuing...
BOOTP broadcast 1
*** Unhandled DHCP Option in OFFER/ACK: 125
*** Unhandled DHCP Option in OFFER/ACK: 125
DHCP client bound to address 192.168.1.14 (229 ms)
*** Warning: no boot file name; using 'C0A8010E.img'
Using RTL8169#0 device
TFTP from server 192.168.1.1; our IP address is 192.168.1.14
Filename 'C0A8010E.img'.

gallit, 2015-11-12:
I think i missed something.
I did a [set test "toto"] and tried to boot with no success.
So i guess it's not an environment variable that is tested here.

Gnurou, 2015-11-12:
Mmmm, for some reason U-boot is trying to load the 64-bit kernel? (which does not exist and will not run on TK1 anyway).

What is the content of /boot/boot-nouveau.scr.txt on your ARM filesystem?

Gnurou, 2015-11-12:
Ah, sorry - I missed the first part of your message. Something seems to be wrong with the generated script, it should read like this:

    if test $cpu = "armv8"; then
    ....

I screwed up with bash variables - will fix it tomorrow and post again. Sorry that you have to report all these mistakes of mine (I always network-boot so I don't get to test this script)

gallit, 2015-11-12:
Never mind, i'm glad to help a bit.

Gnurou, 2015-11-13:
This should be fixed now - you will need to run ./scripts/build-linux again to regenerate the U-boot script and copy /boot again to your SD card. Can you confirm this is working now?

gallit, 2015-11-13:
Great ! It's working good now.

closed 2015-11-13 by gallit