NVIDIA/tegra-nouveau-rootfs #20: weston / xorg is not starting

gallit, 2015-11-13:
I don't have weston-launch working.
I dont know how to log ma tty output.
I tried to "> toto.log" and using "script", no success.
So i took a picture : http://ibin.co/2MOXyupYnOcv
Please tell me if there is another way to do.

I also tried xorg : startx does not work.
Log file is here : http://pastebin.com/yf23knAi

I don't know if it's related but i had some warnings during the build process of weston and xorg :
libtool: warning: remember to run 'libtool --finish /opt/nouveau/lib/xorg/modules/input'

Gnurou, 2015-11-16:
Mmm, that's weird. Is "kmscube /dev/dri/card0 /dev/dri/renderD128" working?

gallit, 2015-11-16:
No it's not working. A can't give a a trace for now.

Gnurou, 2015-11-16:
Ok, let's start with kmscube then, as it is the simplest program of all. I'd love to see a log of it when you can, thanks!

gallit, 2015-11-17:
Well, i think the troubles come from my rootfs that is too old. 
I wanted to keep my jetson system configuration so i rsynced in both ways.

I've just rebuilt everything from scratch, it works fine and great.
Sorry for that bad issue report.

gallit, 2015-11-17:
Just for information, the install-firmware is not working.
I guess it's for maxwell firmware, so for armv8 only.

closed 2015-11-17 by gallit

Gnurou, 2015-11-19:
Could you detail what is wrong with install-firmware? It seems to be working just fine for me.

gallit, 2015-11-19:
It's about a folder that does not exists.

[gallit@archi jetson]$ ./scripts/install-firmware 
cp: cannot stat ‘/home/gallit/dev/jetson/linux-firmware.git/nvidia’: No such file or directory

gallit, 2015-11-19:
I don't have this folder when i sync.

[gallit@archi jetson]$ ls -l
total 60
drwxr-xr-x 17 gallit gallit 4096 Nov 17 15:29 drm
drwxr-xr-x  6 gallit gallit 4096 Nov 17 15:38 kmscube
drwxr-xr-x 25 gallit gallit 4096 Nov 17 10:04 linux
drwxr-xr-x 11 gallit gallit 4096 Nov 17 15:29 mesa
drwxr-xr-x  7 gallit gallit 4096 Nov 17 10:04 nouveau
drwxr-xr-x  5 gallit gallit 4096 Nov 17 13:48 out
drwxr-xr-x  6 gallit gallit 4096 Nov 17 13:46 pkg-config-0.28
drwxr-xr-x  4 gallit gallit 4096 Nov 17 15:29 pthread-stubs
drwxr-xr-x  4 gallit gallit 4096 Nov 17 10:04 scripts
drwxr-xr-x 11 gallit gallit 4096 Nov 17 13:46 wayland
drwxr-xr-x 18 gallit gallit 4096 Nov 17 15:38 weston
drwxr-xr-x  9 gallit gallit 4096 Nov 17 15:41 xf86-input-libinput
drwxr-xr-x  5 gallit gallit 4096 Nov 17 15:41 xkbcomp
drwxr-xr-x 16 gallit gallit 4096 Nov 17 15:41 xkeyboard-config
drwxr-xr-x 32 gallit gallit 4096 Nov 17 15:39 xserver

Gnurou, 2015-11-19:
Ah, of course - this repo is only checked out for 64-bit systems. Just fixed it, thanks!

gallit, 2015-11-19:
You fixed it ?
I still dont have the folder after syncing.

Gnurou, 2015-11-19:
You don't need that folder. install-firmware is a no-op on 32-bit systems.

gallit, 2015-11-19:
Ok