NVIDIA/tegra-nouveau-rootfs #24: build errors in kmscube, weston etc

pushkar90, 2016-02-22:
86_64_aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/5.2.1/../../../../aarch64-linux-gnu/bin/ld: cannot find -lEGL
/aarch64-linux-gnu/bin/ld: cannot find -lGLESv2
etc

Can you tell me where I'm going wrong? Afaik it pulls the needed files from git, but maybe there's a problem with symbolic links?

Gnurou, 2016-02-22:
Could you post a more complete log of your build attempt on pastebin and post the link here?

pushkar90, 2016-02-22:
Went to bed. But I managed to solve the EGL related issues while building weston.
The tegra-nouveau-rootfs directory had a bunch of broken links. So I found the appropriate files it's supposed to link to and added them in.

Now I get this error? after running build-xserver. Xserver and libinput succesfully installed, I just don't understand what it's trying to do right after that or if it's important at all.

http://pastebin.com/gLun0iwA

Gnurou, 2016-02-23:
I think at this stage the best thing to do would be to wipe out the out/ directory, re-run "repo sync -c", and run the scripts again from the beginning. You're probably not doing anything wrong but maybe you built a newer version on top of an older one and that caused these issues.

Please let me know if you are still having these issues on a fresh build.

zeroepoch, 2016-04-19:
May be related to realpath issues I created a ticket for since it creates some broken symlinks.  I never ran into this issue starting from scratch today with coreutils updated to the version of vivid.

Gnurou, 2016-04-19:
Very likely. I have commited a fix to #27 , so closing this issue.

closed 2016-04-19 by Gnurou