NVIDIA/tegra-nouveau-rootfs #25: Build error on Ubuntu 14.04

arm000, 2016-04-11:
Following the instructions for installing required packages on Ubuntu 14.04:

`$ sudo apt-get install git build-essential wget proot phablet-tools autoconf automake libtool libc6-i386 lib32stdc++6 lib32z1 pkg-config libwayland-dev bison flex bc u-boot-tools glib-2.0 realpath libffi-dev`

Installs version 3.0.2 of proot:

```
arm@chrome~/tegra-nouveau-rootfs$ proot --version
PRoot 3.0.2: chroot, mount --bind, and binfmt_misc without privilege/setup.
built-in accelerators: process_vm = yes, seccomp_filter = yes
Visit http://proot.me for help, bug reports, suggestions, patchs, ...
Copyright (C) 2013 STMicroelectronics, licensed under GPL v2 or later.
```

Which does not support the -S option:

```
arm@chrome~/tegra-nouveau-rootfs$ ./scripts/prepare-rootfs 
Setting current user as owner of target FS... 
Using qemu to install required packages on target FS... 
proot error: unknown option '-S'.
proot error: see `proot --help` or `man proot`.
```

Building proot from TOT source fixes the issue (5.0.0-24)

zeroepoch, 2016-04-18:
I'm seeing the same issue on 14.04 with everything up to date.  I'll try the latest proot as suggested by arm000 which should hopefully fix the problem.

zeroepoch, 2016-04-18:
I actually found a simpler solution.  I just downloaded and installed the proot package from vivid which is 5.1.0 and that seems to work.

Gnurou, 2016-04-19:
It seems that we could replace the -S option with -0 -r and a few -b. I will try to find a working combo.

Gnurou, 2016-04-19:
This should be fixed by commit 4d9f4e500bd8. Thanks for reporting!

closed 2016-04-19 by Gnurou