NVIDIA/tegra-nouveau-rootfs #27: Build error compiling weston on Ubuntu 14.04

arm000, 2016-04-11:
Following build instructions in README for Ubuntu 14.04, building weston fails, because it cannot find the GL libraries

```
arm@chrome~/tegra-nouveau-rootfs$ ./scripts/build-weston
...
  CCLD     vertex-clip.test
  CCLD     matrix-test
  CCLD     setbacklight
/home/arm/tegra-nouveau-rootfs/out/host/gcc-linaro-4.8-2015.06-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.8.5/../../../../arm-linux-gnueabihf/bin/ld: warning: libEGL.so.1, needed by /home/arm/tegra-nouveau-rootfs/out/target/arm/ArchLinuxArm/lib/libcairo.so, not found (try using -rpath or -rpath-link)
/home/arm/tegra-nouveau-rootfs/out/host/gcc-linaro-4.8-2015.06-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/4.8.5/../../../../arm-linux-gnueabihf/bin/ld: warning: libGL.so.1, needed by /home/arm/tegra-nouveau-rootfs/out/target/arm/ArchLinuxArm/lib/libcairo.so, not found (try using -rpath or -rpath-link)
/home/arm/tegra-nouveau-rootfs/out/target/arm/ArchLinuxArm/lib/libcairo.so: undefined reference to `glXMakeCurrent'
/home/arm/tegra-nouveau-rootfs/out/target/arm/ArchLinuxArm/lib/libcairo.so: undefined reference to `glEnable'
...
```

This is because the library symlinks are pointing back to build host directories:
```
arm@chrome~/tegra-nouveau-rootfs$ ls -l out/target/arm/ArchLinuxArm/usr/lib/libEGL.*
lrwxrwxrwx 1 root root 15 Apr 10 11:17 out/target/arm/ArchLinuxArm/usr/lib/libEGL.so -> libEGL.so.1.0.0
lrwxrwxrwx 1 root root 15 Apr 10 11:17 out/target/arm/ArchLinuxArm/usr/lib/libEGL.so.1 -> libEGL.so.1.0.0
lrwxrwxrwx 1 root root 29 Apr 10 11:17 out/target/arm/ArchLinuxArm/usr/lib/libEGL.so.1.0.0 -> /usr/lib/mesa/libEGL.so.1.0.0
arm@chrome~/tegra-nouveau-rootfs$ ls -l out/target/arm/ArchLinuxArm/usr/lib/libGL.*
lrwxrwxrwx 1 root root 14 Apr 10 11:17 out/target/arm/ArchLinuxArm/usr/lib/libGL.so -> libGL.so.1.2.0
lrwxrwxrwx 1 root root 14 Apr 10 11:17 out/target/arm/ArchLinuxArm/usr/lib/libGL.so.1 -> libGL.so.1.2.0
lrwxrwxrwx 1 root root 28 Apr 10 11:17 out/target/arm/ArchLinuxArm/usr/lib/libGL.so.1.2.0 -> /usr/lib/mesa/libGL.so.1.2.0
```
Fixing up the symlinks to be relative resolved the issue:

```
arm@chrome~/tegra-nouveau-rootfs/out/target/arm/ArchLinuxArm/usr/lib$ sudo ln -f -s mesa/libGL.so.1.2.0 .
arm@chrome~/tegra-nouveau-rootfs/out/target/arm/ArchLinuxArm/usr/lib$ sudo ln -f -s mesa/libEGL.so.1.0.0 .

```

zeroepoch, 2016-04-18:
I think this has been resolved, although it leads to another issue which is realpath uses the -m option which I've already reported a bug for (aka missing steps in preparation on 14.04)

zeroepoch, 2016-04-19:
This looks like the same issue as #24 ?

Gnurou, 2016-04-19:
I think @zeroepoch is right - realpath failed to run and thus the absolute links were not replaced with relative ones. I have just pushed a fix to that - if you run prepare-rootfs again, I'd bet things will get better.

Note that if you update the target FS on the board (using pacman -Syu), you may need to run prepare-rootfs again to fix the links if they have been updated. There are only a handful of packages using absolute links for libraries, but sadly there is not much we can do to address this besides hot-fixing them...

closed 2016-04-19 by Gnurou