NVIDIA/tegra-nouveau-rootfs #28: Missing -m option on realpath

zeroepoch, 2016-04-18:
An up-to-date Ubuntu 14.04.4 LTS includes coreutils 8.21 which is missing the -m option on the realpath command.  This results in all the symlinks for the various libraries created by prepare-rootfs to be in the trunk directory rather than the correct place.  I fixed the problem by install coreutils 8.23 from vivid.  Maybe there should be some mention in the README.md in the host preparation section about requiring proot and coreutils from vivid.

Gnurou, 2016-04-19:
Just pushed a change that does not require -m. Now the realpath on Ubuntu 14.04 should work just fine. Thanks!

zeroepoch, 2016-04-21:
I don't know if I can reopen this issue, but `realpath` on Ubuntu 14.04 is very basic and doesn't support `--relative-to` either.  I'm not sure there is any easier solution than upgrading coreutils.

```
$ realpath --help
Usage:
 realpath [-s|--strip] [-z|--zero] filename ...
 realpath -h|--help
 realpath -v|--version
```

Gnurou, 2016-04-22:
Let's reopen and see if I can workaround this as well.

Gnurou, 2016-04-22:
Found out about the -r option of ln, which appears to be available in Ubuntu 14.04! This makes realpath unneeded, so hopefully issue fixed.

closed 2016-04-22 by Gnurou

zeroepoch, 2016-04-22:
Just a minor cleanup but you can remove `realpath` from the list of Ubuntu dependencies.

Gnurou, 2016-04-22:
Ah right, done. If you get a chance to test this, please report your result as I haven't tested yet.

zeroepoch, 2016-04-22:
I was just going to but realized the system I used for doing the builds is set right now to boot into Windows.  I'll have to wait until tomorrow I guess so I can change the boot option from grub to Ubuntu manually.

Gnurou, 2016-04-22:
No worries. This is a very trivial change so I expect it to be fine - I will have a chance to test myself later today, so don't rebuild a system just for that :)

zeroepoch, 2016-04-22:
It's a dual-boot system so I just need to boot into Ubuntu, resync and run the scripts.  Not a big deal but you may get to it before me.  It's at the office which is why I probably won't get a chance until tomorrow PDT.

zeroepoch, 2016-04-23:
Turns out there is a bug in `ln` for the version of `coreutils` used by Ubuntu that has been fixed already so not a problem for Arch.  It follows the target link no matter what and mseses up the relative path.  Ubuntu is just all sorts of fun problems.  Anyways decided to just give a patch rather than doing a pull request and all that since it's a simple change.

```
From 5fe617aa6524910e6f9b6c42b287f614d980c8c9 Mon Sep 17 00:00:00 2001
From: Eric Work <work.eric@gmail.com>
Date: Fri, 22 Apr 2016 23:46:19 -0700
Subject: [PATCH] Workaround ln -r bug in coreutils used by Ubuntu 14.04

---
 distro/prepare-ArchLinuxArm | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/distro/prepare-ArchLinuxArm b/distro/prepare-ArchLinuxArm
index e50ed34..03797ff 100644
--- a/distro/prepare-ArchLinuxArm
+++ b/distro/prepare-ArchLinuxArm
@@ -50,9 +50,9 @@ LINKS=$(find $SYSROOT/usr/lib -type l -not -path "*/usr/lib/modules/*")
 for LINK in $LINKS; do
     L=$(readlink $LINK)
     if [[ "$L" = /* ]]; then
-        R=$(dirname $LINK)
-	ABS="${SYSROOT}${L}"
+        ABS="${SYSROOT}${L}"
         echo Fixing $LINK...
+        rm -f $LINK
         ln -sfr $ABS $LINK
     fi
 done
-- 
1.9.1

```

Gnurou, 2016-04-24:
Awesome, thanks for the patch! Just applied it.