NVIDIA/tegra-nouveau-rootfs #29: Missing dependency python-mako

zeroepoch, 2016-04-18:
You need to install `python-mako` on Ubuntu 14.04 in order to finish the step `./scripts/build-mesa`.  Maybe this is a new requirement?

Gnurou, 2016-04-19:
Indeed, it is a requirement - it was actually listed for Arch Linux. I do not really keep up-to-date with Ubuntu requirements since I am not using it, so you reporting these errors is quite helpful. Thanks.

Incidentally, I think you don't even need to compile libdrm and Mesa anymore since the upstream packages now contain all the necessary patches. I will confirm and update the README accordingly.

closed 2016-04-19 by Gnurou

zeroepoch, 2016-04-19:
I haven't had a chance to test booting the image I created (will do so tomorrow), but with this and a few other changes I reported it successfully builds each of the steps in the README as of today.  Cool that more of the changes are going upstream.

Gnurou, 2016-04-19:
Yeah, eventually we want these scripts to become unneeded, excepted maybe for the people who want to easily cross-compile and boot over a network share.

Gnurou, 2016-04-19:
Oops, correction: libdrm still needs to be compiled due to an extra option that is not enabled on Arch Linux. Stock Mesa is fine though.

commit d90ba247 (2016-04-19): README: add python-mako as dependency for Ubuntu

Fixes #29

Change-Id: Ia94da0ab05b5fb20608b1c8c090e077ab93671d2