NVIDIA/tegra-nouveau-rootfs #31: Fails to load DTB

zeroepoch, 2016-04-21:
Everything appears to have been built successfully and at first I created `extlinux.conf` but later found out you can use the created `boot.scr` instead as-is.  Using either approach I keep running into problems loading the DTB.  The files in `/boot/dtb-nouveau` appear to be DTB files according to the `file` command so I'm not sure what is wrong with them.  Maybe something has changed in the kernel that makes these DTB files incomplete?  See the errors below.

```
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scr
706 bytes read in 105 ms (5.9 KiB/s)
## Executing script at 90000000
12576256 bytes read in 612 ms (19.6 MiB/s)
51116 bytes read in 109 ms (457 KiB/s)
## Flattened Device Tree blob at 82000000
   Booting using the fdt blob at 0x82000000
   Using Device Tree in place at 0000000082000000, end 000000008200f7ab
Can't create DT prop nvidia,emc-shadow-regs-rdwr-train to copy
Can't create DT prop nvidia,current_dram_clktree_c0d0u0 to copy
Can't create DT node emc-table@204000 to copy
ERROR: board-specific fdt fixup failed: FDT_ERR_NOTFOUND
 - must RESET the board to recover.

FDT creation failed! hanging...### ERROR ### Please RESET the board ###
```

Gnurou, 2016-04-21:
Ah, this is an issue with the L4T bootloader. I think I have a fix for this, let me try it and come back to this...

Gnurou, 2016-04-21:
Ok, the fix to this should have hit the repo - please repo sync and run the build-linux script again. You should see the message saying "Writing U-boot script in /boot/boot-nouveau.scr...".

Check on the target filesystem that this file now contains the following line:

    set fdt_copy_node_names

The L4T bootloader needs to propagate a few nodes from the bootloader's DT to the kernel DT. The nodes to copy are controlled by this variable, and if the targets are not found the BL will complain and abort as you see. Resetting this variable skips the copy phase, and Linux should then boot without any issue.

Closing this issue - please reopen if needed!

closed 2016-04-21 by Gnurou

zeroepoch, 2016-04-21:
Yep that worked!  I actually just looked at the diff, changed the boot script, and manually rebuilt the boot.scr file.  Thanks for the quick response.  I'm at the login prompt now so need to continue on to see if the driver is working, but step 1 is boot the kernel :+1:

Gnurou, 2016-04-21:
Excellent, thanks for confirming! At this point if you can see /dev/dri/renderD128 this means that Nouveau is most probably working properly.