NVIDIA/tegra-nouveau-rootfs #33: Keyboard / mouse input no longer working (X11/VT)

zeroepoch, 2016-04-22:
Tested using a fresh build on 2016-04-21 USB keyboards and mice are not working with X11 or virtual terminals.  The problem is more basic, probably related to libinput, and even `evtest` is not working.  It shows the keyboard as 2 devices, 1 for the main keyboard, and 1 for the multimedia keys.  When either is selected for testing no key output is shown as you would expect.  On Fedora the same keyboard works using `evtest` and it show keypresses.

Gnurou, 2016-04-26:
You mention they are not working on X and virtual terminals - how about the FB console? I.e. can you log into and use TTY1? Are the USB devices properly enumerated by the kernel, without errors?

zeroepoch, 2016-04-26:
No, I can not login into tty1 either. The keyboard shows up in lsusb from the serial console and also in evtest. This is a TX1 that is also running R23.2 on eMMC. Not sure what to try next.

Gnurou, 2016-04-26:
That's extremely weird, I cannot reproduce this at all on a fresh setup. Can you maybe create a tarball of the produced filesystem and put it on a file sharing site (GDrive or anything else) so I can test it and see the differences with mine?

zeroepoch, 2016-04-27:
Sure. I'll get it to you when I'm back. I'm out of the country at the moment. I wonder if it had anything to do with me using Ubuntu instead of Arch.

Gnurou, 2016-04-27:
As far as the host is concerned this should not be a problem. If you mean you are using Ubuntu on the target filesystem, then yeah I would understand things do not work!

zeroepoch, 2016-04-27:
I was referring to using Ubuntu as the build system and using Arch as the target.  I wonder if certain versions of the build tools being older causes problems. Maybe something gets disabled. Once I get a chance to get you a copy of the rootfs I guess we'll know.

Gnurou, 2016-04-27:
Thanks. Maybe one of the scripts is not running properly due to the version difference in tools? Having the rootfs will allow me to diff against mine and see what's different.

Gnurou, 2016-05-11:
Just booted on the rootfs you sent me, and USB worked just fine both under TTY1 and X. Both keyboard and mouse worked. That's extremely strange. Can you try with a different keyboard maybe? And with a powered USB hub between the board and the devices?

zeroepoch, 2016-05-11:
I didn't try a hub but I'll try that and also a different keyboard. Thanks for testing.

zeroepoch, 2016-05-17:
I still haven't had the chance to grab another keyboard or hook up a USB hub, but I did remove the SD card and let it boot to L4T both to X11 and with lightdm disabled.  The same keyboard works in both cases.  As soon as I reboot with the SD card inserted I can't type in the terminal on-screen anymore.  Since my rootfs seems to work for you maybe it's something with this model of the keyboard and libinput?  When I get a chance to try another model of keyboard I'll add another comment.

Gnurou, 2016-05-17:
Ah, that's a very significant detail, thanks. I will try to boot with a SD card inserted.

In that scenario, do you boot *from* the SD card or does a SD card just happens to be inserted when you boot from eMMC? In the former case, are the rootfs on eMMC and SD card the same?

zeroepoch, 2016-05-17:
I'm booting off the SD card (kernel + rootfs).  The rootfs on eMMC is still the latest L4T.  The filesystem on the SD was what I tar'd and provided to you earlier.  The eMMC should not be involved in terms of filesystem content when booting the Arch Linux image.

Gnurou, 2016-05-18:
Ah ok, so the boot medium is not important here - what matters is that with L4T your input devices work, but not with upstream Arch Linux (I have tried booting with a SD card inserted and things worked just fine for me).

I'd like to see what happens if you use a powered hub, just in case, but what works on L4T should also on upstream. It is just that from your report, the problem seems to come from user-space software and I would like to confirm that.

The output of lsusb and dmesg when the (non-working) keyboard is plugged would also be helpful.

zeroepoch, 2016-05-20:
Sorry it took so long to get back to you.  I tried a different keyboard and that DIDN'T work.  I tried a powered USB 3.0 hub and that DID work.  I tried removing the hub and it DIDN'T work anymore.  In fact both keyboards work at the same time but the only working combination is with the keyboards behind the USB hub.

See the info requested below:

```
Bus 002 Device 002: ID 0955:09ff NVidia Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 413c:2112 Dell Computer Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

http://pastebin.com/3hTUWdr8

Gnurou, 2016-05-25:
Ok, so the problem seems to be that some non-powered USB devices are not working properly on upstream, while they are performing just well on L4T.

@thierryreding , is this a known issue with USB on upstream TX1? My own keyboard (HKKB) seems to work just fine without a usb hub...

thierryreding, 2016-05-25:
Did you mean to say TK1? USB shouldn't currently work at all on TX1. None
of the boards I know of have the EHCI connected to anything, so I think
you'd absolutely need the XUSB support for Tegra210 and that's not merged
upstream yet.

I'm not aware of any problems with USB on TK1, though I certainly haven't
tested a very broad spectrum of devices.

zeroepoch, 2016-05-25:
This is on the TX1.  I actually have a TK1 as well so I did a build for that platform and as expected the USB keyboard works with the USB hub.  From what @thierryreding is saying it's not expected to work although somehow the TX1 does work with a USB hub.

Gnurou, 2016-05-25:
@thierryreding this is TX1 indeed, but on your staging/work branch - not pure upstream actually.

Since it has just been updated (yay!) we should probably try again with the new head?

zeroepoch, 2016-11-30:
Closing out this old ticket since I am no longer working with this board and won't have time to update and retest this.

closed 2016-11-30 by zeroepoch