NVIDIA/tegra-nouveau-rootfs #35: uboot broken after flash

pfc, 2017-05-17:
After flashing the system over USB and restarting I get the following:

[0000.153] [TegraBoot] (version 24.00.2015.42-mobile-ec3b827e)
[0000.158] Processing in cold boot mode Bootloader 2
[0000.163] A02 Bootrom Patch rev = 63
[0000.166] Power-up reason: software reset
[0000.170] No Battery Present
[0000.173] RamCode = 0
[0000.175] Platform has Ddr4 type ram
[0000.178] max77620 disabling SD1 Remote Sense
[0000.183] Setting Ddr voltage to 1125mv
[0000.186] Serial Number of Pmic Max77663: 0x501b7
[0000.194] Entering ramdump check
[0000.197] Get RamDumpCarveOut = 0x0
[0000.200] RamDumpCarveOut=0x0,  RamDumperFlag=0xe59ff3f8
[0000.205] Last reboot was clean, booting normally!
[0000.210] Sdram initialization is successful 
[0000.214] SecureOs Carveout Base=0xff800000 Size=0x00800000
[0000.220] GSC1 Carveout Base=0xff700000 Size=0x00100000
[0000.225] GSC2 Carveout Base=0xff600000 Size=0x00100000
[0000.230] GSC3 Carveout Base=0xff500000 Size=0x00100000
[0000.235] GSC4 Carveout Base=0xff400000 Size=0x00100000
[0000.240] GSC5 Carveout Base=0xff300000 Size=0x00100000
[0000.245] BpmpFw Carveout Base=0xff2c0000 Size=0x00040000
[0000.251] Lp0 Carveout Base=0xff2bf000 Size=0x00001000
[0000.266] RamDump Carveout Base=0xff23f000 Size=0x00080000
[0000.271] Platform-DebugCarveout: 0
[0000.275] Nck Carveout Base=0xff03f000 Size=0x00200000
[0000.280] Non secure mode. Disable rollback prevention
[0000.285] AOTAG Init Done
[0000.312] Using GPT Primary to query partitions 
[0000.317] GPT: Partition NOT found ! 
[0000.321] Find Partition via GPT Failed
[0000.324] Find Partition via PT Failed
[0000.328] function NvTbootGetBinaryOffsets: 0x1 error
[0000.333] Error in NvTbootLoadBinary: 0x1 !
[0000.337] GPT: Partition NOT found ! 
[0000.340] Find Partition via GPT Failed
[0000.344] Find Partition via PT Failed
[0000.347] function NvTbootGetBinaryOffsets: 0x1 error
[0000.352] Error in NvTbootLoadBinary: 0x1 !
[0000.356] Error is 1 

I've previously flashed my own copy of uboot using the flash.sh from L4T:
./flash.sh -L ../../uboot09052017/u-boot.bin  jetson-tx1 mmcblk0p1

I'm assuming the step 4 "partition the eMMC to have one single ext4 partition." deletes the flashed uboot.  Should there be another step after 4 to restore uboot?

JoelLisenby, 2017-09-25:
I ran into this same issue. Not quite sure where to go from here.

```
[0000.759] [TegraBoot] (version 24.00.2015.42-mobile-ec3b827e)
[0000.764] Processing in cold boot mode Bootloader 2
[0000.769] A02 Bootrom Patch rev = 63
[0000.772] Power-up reason: on button
[0000.776] No Battery Present
[0000.778] RamCode = 0
[0000.781] Platform has Ddr4 type ram
[0000.784] max77620 disabling SD1 Remote Sense
[0000.788] Setting Ddr voltage to 1125mv
[0000.792] Serial Number of Pmic Max77663: 0x1019ae
[0000.800] Entering ramdump check
[0000.803] Get RamDumpCarveOut = 0x0
[0000.806] RamDumpCarveOut=0x0,  RamDumperFlag=0xe59ff3f8
[0000.811] Last reboot was clean, booting normally!
[0000.816] Sdram initialization is successful 
[0000.820] SecureOs Carveout Base=0xff800000 Size=0x00800000
[0000.826] GSC1 Carveout Base=0xff700000 Size=0x00100000
[0000.831] GSC2 Carveout Base=0xff600000 Size=0x00100000
[0000.836] GSC3 Carveout Base=0xff500000 Size=0x00100000
[0000.841] GSC4 Carveout Base=0xff400000 Size=0x00100000
[0000.846] GSC5 Carveout Base=0xff300000 Size=0x00100000
[0000.851] BpmpFw Carveout Base=0xff2c0000 Size=0x00040000
[0000.857] Lp0 Carveout Base=0xff2bf000 Size=0x00001000
[0000.872] RamDump Carveout Base=0xff23f000 Size=0x00080000
[0000.877] Platform-DebugCarveout: 0
[0000.881] Nck Carveout Base=0xff03f000 Size=0x00200000
[0000.886] Non secure mode. Disable rollback prevention
[0000.891] AOTAG Init Done
[0000.938] Using GPT Primary to query partitions 
[0000.944] GPT: Partition NOT found ! 
[0000.947] Find Partition via GPT Failed
[0000.951] Find Partition via PT Failed
[0000.954] function NvTbootGetBinaryOffsets: 0x1 error
[0000.959] Error in NvTbootLoadBinary: 0x1 !
[0000.964] GPT: Partition NOT found ! 
[0000.967] Find Partition via GPT Failed
[0000.971] Find Partition via PT Failed
[0000.974] function NvTbootGetBinaryOffsets: 0x1 error
[0000.979] Error in NvTbootLoadBinary: 0x1 !
[0000.983] Error is 1
```

pfc, 2017-09-25:
The solution was to flash uboot and only repartition the one partition. The Jetson will not boot without the following partitions: app tbc bpf ebt rp1 dtb wb0 tos.

jonhunter, 2017-09-25:
Thanks for the update. Let me know if you need anything further I will try to get to it in the next week or so.

JoelLisenby, 2017-09-26:
Thanks! I reinstalled the stock image, then repartitioned the drive this time only deleting/recreating partition 1 and then formatting partition 1 to ext4. Booting up now :)